armel/uv-k5-firmware-custom-feat-F4HWN #77: Suggestion pour le One Push

F6GDN, 2024-05-06:
Bonjour Armel !

Merci encore pour ton excellent travail et les versions de firmware qui s'enchaînent à un rythme soutenu.
Tu dois y passer des nuits... Hi :-)
Je n'ai pas réussi à reproduire le bug signalé dans mon message précédent, je suis perplexe :-(
En revanche, j'ai trouvé pourquoi je ne pouvais pas relancer le scan, tout simplement parce que j'étais en mode One Push.
Donc, j'arrêtais le scan en appuyant sur le PTT, mais je ne pouvais pas le relancer tant que je n'avais pas pressé à nouveau sur le PTT, ce qui est logique :-).

Justement, concernant le mode One Push, comme une simple pression furtive sur le PTT passe le poste en émission, et le bloque dans ce mode, une simple manutention inattentive peut provoquer une émission involontaire, et bloquer le micro ouvert.

Serait-il possible de valider le One Push par une pression plus longue sur le PTT, par exemple 2 secondes, ce qui confirmerait un acte volontaire en émission ?

Merci encore et tous mes encouragements !

73 !
Eddie
F6GDN

armel, 2024-05-07:
Bonjour Eddie,

Merci pour ce message.

Hummm, ce serait compliqué de baser la validation OnePush sur une pression d'une durée déterminée, genre 2 secondes. Ca voudrait dire qu'en pratique, je fais un appui prolongé sur le PTT, je compte au moins 2 secondes dans ma tête et... seulement après je commence à parler ? Ca ne me semble pas viable. 

Dans les faits, il faut garder en tête que si le OnePush apporte un confort énorme, il impose en retour la plus grande vigilance. Personne n'est effectivement à l'abris de passer en émission par inadvertence. Il faut juste le savoir.

Armel F4HWN.

F6GDN, 2024-05-07:
Bonjour Armel !

Merci beaucoup pour cette réponse rapide.
Tu as raison, le One Push apporte un confort énorme.
Dans mon idée, c'était que le PTT passe en mode One Push lorsque l'on le
maintient appuyé plus de 2 secondes, par exemple.
Mais en effet, à nous de garder un oeil sur la LED rouge pour éviter de
passer en émission sur fausse manipulation ;-)

Je te souhaite une excellente journée.
Cordiales 73 !
Eddie
F6GDN

armel, 2024-05-07:
Bonjour Eddie,

Oui, la LED rouge est là pour alerter sur les passages en transmission. 
Je reste conscient qu'il y a un risque d'erreur de manip. J'en avais même parlé en présentant cette fonctionnalité la première fois. Mais c'est un coup de main à prendre, comme à chaque prise en main d'une nouvelle fonctionnalité dont on a pas encore l'habitude :) 

Je cloture l'issue.

Excellente journée à toi et à tes proches.

73'

Armel F4HWN.

closed 2024-05-07 by armel