armel/uv-k5-firmware-custom-feat-F4HWN #78: V2.6 réinitialisation du k5 quand extinction

14UVR10, 2024-05-06:
Les réglages genre Switch mode ,ou W/N,.. Sont remis à zéro après rallumage du Tx
Genre : a l'extinction, je suis en mode MO, après allumage DWR
Narrow, revient en Wide après allumage  

Lien du pdf 2.7
[20900302](https://www.dropbox.com/scl/fi/um2oaupg7c7zx8ug9fpdd/MENU-Armel-F4HWN-2.7.pdf?rlkey=7294nxdpwgpovuz1l53jxd79h&e=2&dl=0)

armel, 2024-05-06:
Tout est normal. 

Les changements apportaient via raccourcis de type SWITCH RxMode ou SWITCH WIDE NARROW sont temporaires. L'idée est de faire un changement rapide, à la volée, sans passer par les menus. 

Pour changer de mode d'affichage et faire en sorte qu'il persiste à l'allumage, il faut utiliser le menu 53/63 RxMode.

Pour changer la largeur de bande de Wide à Narrow (ou inversement) et faire en sorte qu'il persiste, il faut également penser à confirmer le changement via le menu 15/63 ChSave. Les informations comme la puissance, la largeur de bande, le type de modulation, les subaudibles (CTCSS, DCS, etc.) sont dit "_per channel_", c'est à dire qu'ils dépendent de chaque canaux mémoire, de façon indépendante. D'ou la nécessité de confirmer les changements par le menu 15/63 ChSave pour les garder en mémoire et les retrouver à l'allumage. 

73'

Armel F4HWN.

14UVR10, 2024-05-07:
ok , je transmets au groupe FB Merci pour ta réponse 73

armel, 2024-05-07:
Merci pour le QSP vers le groupe FB, c'est très gentil de ta part. Je ferais peut-être une vidéo, à l'occasion, sur ce genre de questions. Je cloture cette issue. 73'

Armel F4HWN

closed 2024-05-07 by armel