armel/uv-k5-firmware-custom-feat-F4HWN #79: Which file to flash?

troydm, 2024-05-08:
Hi, there are two files in this repository, one is called f4hwn.packed.bin and another is firmware.packed.bin, which one should I flash and what is the difference between them?

armel, 2024-05-08:
Hello,

These are exactly the same. At the beginning of my developments, several months ago, the binary was called firmware.packed.bin. It was too generic. Since upgrading to version 2, I call it f4hwn.packed.bin. But there's no difference between the 2.

By the way, there's really no question to ask. If you click on the link to the [online flasher](https://egzumer.github.io/uvtools/?firmwareURL=https://github.com/armel/uv-k5-firmware-custom-feat-F4HWN/raw/main/f4hwn.packed.bin) (indicated in the README), it will flash the f4hwn.packed.bin version.

Armel F4HWN

troydm, 2024-05-08:
thank you for the answer

closed 2024-05-08 by troydm